MIPT-ILab/mipt-mips #643: Purge `static_cast`s from our code

pavelkryukov, 2018-10-13:
Bjarne Stroustrup's advise not to use `static_cast` for arithmetic types ([link](https://github.com/isocpp/CppCoreGuidelines/blob/master/CppCoreGuidelines.md#prosafety-type-safety-profile)). Instead, ctors should be called:

```c++
 auto f = static_cast<int>(v); // avoid
 int f(v); // prefer;

 x += static_cast<uint32>(y); // avoid
 x += uint32(y); // prefer
```

davtyan-arsen, 2018-11-16:
I have tried to remove `static_cast`s from the code but I've come across a problem. While some of them can be easily transformed to ctors( e.g. conversion from `unsigned long long int` to `uint64`), the majority fails due to `[-Werror=narrowing]`. My question is: should I change the destination types or there is another way around it?

pavelkryukov, 2018-11-16:
For that cases, we must use `narrow_cast`. I opened a pull request with a simple implementation — just a wrapped around `static_cast`. Later, we'll add dynamic check that value does not lose anything due to narrowing.

commit b562c7c4 (2018-11-16): Purge arithmetic `static_cast`s from the code

Fixes #643
closed 2018-11-16 by pavelkryukov