MIPT-ILab/mipt-mips #668: Using `Simulator` instances in unit tests causes duplicating config keys

trexxet, 2018-10-22:
I'm close to finish register interfaces, however, I'm facing an unexpected issue. 

Register interfaces are implemented in `Simulator` class, which relies on some config keys, so unit test for interfaces use `Simulator` instance. Unit test for config parser uses the same key names ("b", "d" etc) as `Simulator` do. Because of this, POPL throws an exception for duplicating keys. 

I guess that `Simulator` instances were not used in unit tests before, so compiler removed that keys during optimization. 

POPL holds all the keys global. 

**To Reproduce**
Add a unit test with something like `auto sim = Simulator::create_simulator ("mips32", true, false"); compile; run.
gcc 7.3.0, Kali Linux (Debian based)

trexxet, 2018-10-22:
At this moment, I can suggest renaming keys used in config parser unit test to values not used by simulator. In this case, this unit test won't simulate real simulator usage, however, it tests config parser routines, not the keys themselves.

pavelkryukov, 2018-10-22:
See PR #669. I keep -b and -n as they live in main.cpp and should not interfere.
Refactoring architecture of configuration values is in my long-term to-do list (I want to keep configs locally to code of configured module and out of static scope, which is a challenge).

pavelkryukov, 2018-10-22:
For now, you may include `func_sim.h` directly to your tests.

closed 2018-10-23 by pavelkryukov