MIPT-ILab/mipt-mips #723: Inherit MIPSInstr and RISCVInstr from a common data structure

pavelkryukov, 2018-11-26:
You should implement a `func_sim/instr.h` header containing a following data structure:

```c++
template<typename Register, typename R>
struct Instr
{
    using RegisterUInt = R;
    Register src1;
    Register src2;
// ...
    RegisterUInt v_src1;
    RegisterUInt v_src2;
// ..
    Operation operation;
    Trap trap;
    Addr mem_addr;
    Addr mem_size;
    RegisterUInt mask;
// ...
};
```

It should contain all data fields which are independent from ISA. Then, derived structure should contain only ISA-specific fields:

```c++
template<typename R>
struct MIPSInstr : Instr<MIPSRegister, R>
{
    const uint32 raw;
    KryuCowString disasm;
// ...
};
```

All the getters/setters methods should be removed if the field is public now.

agrachiv99, 2018-11-29:
I don't fully understand how and where these new structures should be used, can you please explain?

pavelkryukov, 2018-11-29:
At the moment BaseMIPSInstr and RISCVInstr are used to model data path. For example, in FuncSim:

https://github.com/MIPT-ILab/mipt-mips/blob/0f4ab8f9f362b5ecbc2d35591ca1de975df31f3b/simulator/func_sim/func_sim.cpp#L31-L39 

FuncInstr here is a type name deduced from template parameter "ISA". For MIPS32, it is MIPS32Instr, for RISC-V, it is RISCVInstr. If you look at both structures, they contain exactly the same fields — and we need to use a common base class to use all the benefits of C++ inheritance.

pavelkryukov, 2019-02-23:
@agrachiv99 Do you have any more questions?

commit 50c73c32 (2019-02-26): Fix #723: inherit MIPSInstr and RISC-VInstr from common ancestor (#868)
closed 2019-02-26 by pavelkryukov