MIPT-ILab/mipt-mips #724: Extract BrandMispredictDetection unit out of Mem module to a separate module

pavelkryukov, 2018-11-27:
Currently Mem module handles two operations:
1. Loads and stores to/from data memory
2. Branch conditions check.

Actually mem should check only the first one, while the second one should be handled in a separate module.

vodogrey2012, 2018-11-28:
Hello. I am going to try to solve this issue at the closest weekend. But now I have any common questions:
1) As I understand, I need to create new method. It means, I need to make new header file, wich contains one function branch_predictor and its cpp-file with realisation? 
2) This header must be uncluded in mem module and brunc_predictor must be called in clock function, is it?
3) How can I test this modul?
Thank you.

pavelkryukov, 2018-11-29:
> As I understand, I need to create new method.

No, it should be a separate class. Currently we have 6 modules located in `simulator/modules`:
* core (includes other modules)
* fetch
* decode
* execute
* mem
* writeback

The new module should be the 7th one. I advise you to start with copying Mem module and removing everything unrelated.

> This header must be uncluded in mem module and brunc_predictor must be called in clock function, is it?

As the module is same-rank as mem, it should be called from core module clock:
https://github.com/MIPT-ILab/mipt-mips/blob/0f4ab8f9f362b5ecbc2d35591ca1de975df31f3b/simulator/modules/core/perf_sim.cpp#L76-L84

> How can I test this modul?

Run your bubble sort trace, for instance — it should demonstrate performance about 0.6 IPC if everything is correct. Additionally, you may create unit tests, but that's tricky with ports at the moment.

commit 21c5f8f4 (2018-12-04): Add PREDICTION module (#731)

Fixes #724
closed 2018-12-04 by pavelkryukov