MIPT-ILab/mipt-mips #821: Integrate dense_hash_map

pavelkryukov, 2019-02-08:
`std::unordered_map` implementation is quite slow. Instead, we should use faster `dense_hash_map` container: https://github.com/sparsehash/sparsehash-c11.

Your objective is to copy headers and replace `std::unordered_map` with `dense_hash_map` for CacheTagArray classes. `mips_isa.cpp` may stay as is for now since its performance is not so critical.

agrachiv99, 2019-04-19:
Is it possible to measure somehow if after integrating cache started to work faster?

pavelkryukov, 2019-04-19:
Yep. We have a pull request here: https://github.com/MIPT-ILab/mipt-mips/pull/820
It enables profiling information on Travis.
The latest report is here: https://travis-ci.com/MIPT-ILab/mipt-mips/jobs/193620397

```
time   seconds   seconds    calls   s/call   s/call  name    
 25.91      1.50     1.50 12000034     0.00     0.00  Decode<BaseMIPSInstr<unsigned int> >::clock(Cycle)
  6.04      1.85     0.35 44000001     0.00     0.00  ReadPort<PerfInstr<BaseMIPSInstr<unsigned int> > >::read(Cycle)
  5.18      2.15     0.30 22000002     0.00     0.00  InstrMemoryCached<MIPS<(MIPSVersion)7> >::fetch_instr(unsigned long)
  4.66      2.42     0.27 24000002     0.00     0.00  CacheTagArray::read(unsigned long)
  4.49      2.68     0.26 46016380     0.00     0.00  LRUCacheInfo::touch(unsigned long)
  4.40      2.94     0.26 46000004     0.00     0.00  WritePort<PerfInstr<BaseMIPSInstr<unsigned int> > >::basic_write(PerfInstr<BaseMIPSInstr<unsigned int> >&&, Cycle)
  4.23      3.18     0.25 22000000     0.00     0.00  WritePort<std::pair<unsigned int, unsigned int> >::basic_write(std::pair<unsigned int, unsigned int>&&, Cycle)
  3.54      3.39     0.21 110000044     0.00     0.00  BasicWritePort::increment_write_counter(Cycle)
  3.28      3.58     0.19 22000002     0.00     0.00  HierarchiedMemory::memcpy_guest_to_host(Byte*, unsigned long, unsigned long) const
  2.33      3.71     0.14 40000000     0.00     0.00  RF<BaseMIPSInstr<unsigned int> >::write(MIPSRegister, unsigned int, unsigned int, signed char)
  2.25      3.84     0.13 12000004     0.00     0.00  Fetch<BaseMIPSInstr<unsigned int> >::get_target(Cycle)
  1.90      3.95     0.11 12000034     0.00     0.00  Fetch<BaseMIPSInstr<unsigned int> >::clock(Cycle)
  1.90      4.06     0.11 12000034     0.00     0.00  Fetch<BaseMIPSInstr<unsigned int> >::clock_bp(Cycle)
  1.90      4.17     0.11 12000034     0.00     0.00  Execute<BaseMIPSInstr<unsigned int> >::clock(Cycle)
```

pavelkryukov, 2019-04-19:
It may be ran locally, just follow the patch to `.travis.yml`

pavelkryukov, 2019-04-19:
```
  4.66      2.42     0.27 24000002     0.00     0.00  CacheTagArray::read(unsigned long)
  4.49      2.68     0.26 46016380     0.00     0.00  LRUCacheInfo::touch(unsigned long)
```

We're optimizing these.

closed 2019-04-22 by pavelkryukov