MIPT-ILab/mipt-mips #879: Add method to print 128-bit integers in decimal format

pavelkryukov, 2019-03-04:
Currently 128 bit integers cannot be printed as decimal, only least significant 64 bits are exposed:

https://github.com/MIPT-ILab/mipt-mips/blob/3297118b96d5fb768210e5ad6885420d75ae0a8d/simulator/infra/types.h#L52

Your task is to implement or copy-paste 128-bit print along with unit tests.

agrachiv99, 2019-04-17:
I already did it in some contest, if printing it by two halfs is relevant solution. Where should be the unit-test folder? I didn't find any tests for all of the hiders in `infra` folder.

> infra

pavelkryukov, 2019-04-17:
What do you mean by 'printing by 2 halves'?

pavelkryukov, 2019-04-17:
https://github.com/MIPT-ILab/mipt-mips/blob/master/simulator/infra/t/unit_test.cpp

agrachiv99, 2019-04-17:
> What do you mean by 'printing by 2 halves'?

Something like

void print128(uint128 number) {
    if (number <= UINT64_MAX) {
        print64( number);
    } else {
        uint128 leading  = number / 10000000000000000000
        uint64  following = number % 10000000000000000000
        print64( leading);
        print64( following);
    }
}

pavelkryukov, 2019-04-17:
That's hexadecimal, it's simple. Decimal is tricky,

commit 23733b11 (2019-04-22): Switch to the dense_hash_map container (#961)

Add decimal prints of uint128
Fixes #879 and #821
closed 2019-04-22 by pavelkryukov