MIPT-ILab/mipt-mips #901: Implement decoder for compressed RISC-V instructions

pavelkryukov, 2019-03-08:
RISC-V ISA specifies 'C' (compressed) subset of 16-bit encoded instructions: https://content.riscv.org/wp-content/uploads/2017/05/riscv-spec-v2.2.pdf, chapter 12.

Your goal is extended RISCV_Instr_Decoder class to support the compressed instructions, and add these instructions to our RISCV ISA table:

1. C.LWSP
1. C.LDSP
1. C.LQSP
1. C.SWSP
1. C.SDSP
1. C.SQSP
1. C.LW
1. C.LD
1. C.LQ
1. C.J
1. C.JAL
1. C.JR
1. C.JALR
1. C.BEQZ
1. C.BNEZ
1. C.LI
1. C.LUI
1. C.ADDI
1. C.ADDIW
1. C.ADDI16SP
1. C.ADDI4SPN
1. C.SLLI
1. C.SRLI
1. C.SRAI
1. C.ANDI
1. C.MV
1. C.ADD
1. C.AND
1. C.OR
1. C.XOR
1. C.SUB
1. C.ADDW
1. C.SUBW
1. C.EBREAK

BEANefiT, 2019-04-05:
I tried to add some code. Could you please tell me if I am in the right way or not?

pavelkryukov, 2019-04-05:
I propose to start with unit tests first.
We have some tests for RISC-V decoder here:

https://github.com/MIPT-ILab/mipt-mips/blob/3357feb69fee3ee0b27450d8aa02571c24e93641/simulator/risc_v/t/unit_test.cpp#L11-L22

You can write tests by your own, following the ISA paper, or you can use these from LLVM: https://github.com/llvm/llvm-project/blob/master/llvm/test/MC/RISCV/rv32c-valid.s

BEANefiT, 2019-04-12:
I have implemented C_LWSP, C_SW, C_LW instructions on my 'Compressed-ISA' github branch. Could you please have a look at it?

pavelkryukov, 2019-04-12:
I'm on vacation now, please ask Kirill Korolev for code review

pavelkryukov, 2019-04-15:
Please open a pull request to initialize code review tool

commit 103171ef (2019-04-24): Implement decoder for RVC instructions (#960)

Fixes #901
closed 2019-04-24 by pavelkryukov