MIPT-ILab/mipt-mips #910: Port error:     MEMORY_2_EXECUTE_BYPASS has no data to read in cycle:206

pavelkryukov, 2019-03-10:
Steps to reproduce:

```
./mipt-mips -I riscv32 -b .../riscv-tests/isa/rv32ui-p-auipc
```

Output from the last 5 cycles:

```
fetch   cycle 200: 0x80000100: {37}     auipc $a0, 0x2   [ ]
decode  cycle 200: bubble
execute cycle 200: bubble
memory  cycle 200: bubble

wb      cycle 201: bubble
fetch   cycle 201: 0x80000104: {38}     addi $a0, $a0, 1820      [ ]
decode  cycle 201: 0x80000100: {37}     auipc $a0, 0x2   [ ]
execute cycle 201: bubble
memory  cycle 201: bubble

wb      cycle 202: bubble
fetch   cycle 202: 0x80000108: {39}     jal $a1, 4       [ ]
decode  cycle 202: 0x80000104: {38}     addi $a0, $a0, 1820      [ ]
execute cycle 202: 0x80000100: {37}     auipc $a0, 0x2   [ $a0 = 0x80002100 ]
memory  cycle 202: bubble

wb      cycle 203: 0x80000100: {37}     auipc $a0, 0x2   [ $a0 = 0x80002100 ]
fetch   cycle 203: 0x8000010c: {40}     sub $a0, $a0, $a1        [ ]
decode  cycle 203: misprediction on 0x80000108: {39}       jal $a1, 4       [ ]
execute cycle 203: 0x80000104: {38}     addi $a0, $a0, 1820      [ $a0 = 0x8000281c ]
memory  cycle 203: bubble

wb      cycle 204: 0x80000104: {38}     addi $a0, $a0, 1820      [ $a0 = 0x8000281c ]
fetch   cycle 204: 0x8000010c: {40}     sub $a0, $a0, $a1        [ ]
decode  cycle 204: flush
execute cycle 204: 0x80000108: {39}     jal $a1, 4       [ $a1 = 0x8000010c ]
memory  cycle 204: bubble

wb      cycle 205: bubble
fetch   cycle 205: 0x80000110: {41}     lui $t4, 0x2     [ ]
decode  cycle 205: 0x8000010c: {40}     sub $a0, $a0, $a1        [ ]
execute cycle 205: bubble
memory  cycle 205: bubble
0x80000108: {39}        jal $a1, 4       [ $a1 = 0x8000010c ]
wb      cycle 206: 0x80000108: {39}     jal $a1, 4       [ $a1 = 0x8000010c ]
Executed instructions: 40

fetch   cycle 206: 0x80000114: {42}     addi $t4, $t4, 1808      [ ]
decode  cycle 206: 0x80000110: {41}     lui $t4, 0x2     [ ]
execute cycle 206: Port error:  MEMORY_2_EXECUTE_BYPASS has no data to read in cycle:206
```

pavelkryukov, 2019-03-10:
@vodogrey2012 It seems that BRANCH has to implement data bypass logic similarly to the MEMORY stage.

vodogrey2012, 2019-03-10:
Okay, I am going to solve this bug next weekend

vodogrey2012, 2019-03-17:
Goodnight. I ran into an unexpected problem. I'm trying to build a project using Ninja with this line:
```
cmake ../mipt-mips/simulator -G "Ninja"
```
but it shows the following lines:
```
CMake Error at CMakeLists.txt:9 (project):
  Running

   '/usr/sbin/ninja' '--version'

  failed with:

   die: error: must run as root

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "/home/yauheni/programs/mips/bin/CMakeFiles/CMakeOutput.log".
```
but this variables are correctly set in .cmake files.
Also, I have tried to set them using command ``set``, but I get the same result. 
Then, I tried to run this command with `sudo`, but the process freezes without any log.
Also, I tried to use make, but it freezes on 
```
Generating /home/yauheni/programs/mips1/mipt-mips/riscv-opcodes/riscv.opcode.gen.h
```
So, I have no any idea about this problem.
Could you give me any advice about it?
Thank you.

pavelkryukov, 2019-03-18:
How was Ninja installed?

pavelkryukov, 2019-03-18:
Alternatively, you still may use GNU Makefiles: `cmake .. -G "GNU Makefiles && make`

pavelkryukov, 2019-03-18:
> Generating /home/yauheni/programs/mips1/mipt-mips/riscv-opcodes/riscv.opcode.gen.h

Please try to run `git submodule update --init` at the top directory of your workspace. It will create a submodule with RISC-V ISA internals, which we use to autogenerate our code.

vodogrey2012, 2019-03-18:
> How was Ninja installed?

Using ``sudo apt-get install ninja``.
But now I tried to install it with ``sudo apt-get install ninja-build`` and it works, thank you.

vodogrey2012, 2019-03-18:
> > Generating /home/yauheni/programs/mips1/mipt-mips/riscv-opcodes/riscv.opcode.gen.h
> 
> Please try to run `git submodule update --init` at the top directory of your workspace. It will create a submodule with RISC-V ISA internals, which we use to autogenerate our code.

It also helped, thank you.

pavelkryukov, 2019-03-18:
>  I tried to install it with sudo apt-get install ninja-build and it works, thank you.

Yep, I faced the same thing some time ago :-)

closed 2019-04-16 by pavelkryukov