MIPT-ILab/mipt-mips #925: Implement RISC-V M instructions

pavelkryukov, 2019-03-16:
RISC-V ISA has 'M' extension for multiplication and division instructions. We currently decode them, but we do not execute them. Your objective is to implement the executors in https://github.com/MIPT-ILab/mipt-mips/blob/master/simulator/risc_v/riscv_instr.cpp and https://github.com/MIPT-ILab/mipt-mips/tree/master/simulator/func_sim/multiplication.h.

To test instructions, add riscv-tests to FuncSim tests, like the 'I' instructions are added: https://github.com/MIPT-ILab/mipt-mips/blob/1aa8d1d2a585299b94d51b19116cc9bb0271d43b/simulator/func_sim/t/unit_test.cpp#L193

vodogrey2012, 2019-04-17:
Hi. As I see now, this tests were removed in https://github.com/MIPT-ILab/mipt-mips/commit/124f38ad1d7623f01911bb062ccc640ad5d06d0d commit. Should I put it back to [.../risc_v/unit_test.cpp](https://github.com/MIPT-ILab/mipt-mips/blob/124f38ad1d7623f01911bb062ccc640ad5d06d0d/simulator/risc_v/t/unit_test.cpp) ?

pavelkryukov, 2019-04-17:
There are 2 options:
1. Put the M tests back to FuncSim tests, that should work fine.
2. More advanced: copy tests macro from these files https://github.com/MIPT-ILab/riscv-tests-prebuilt/tree/master/isa/rv64um to that file https://github.com/MIPT-ILab/mipt-mips/blob/master/simulator/risc_v/t/riscv64_test.cpp. It is nicer since we do not have to run FuncSim to test implementations, and we do not need prebuilt binary files.

vodogrey2012, 2019-04-17:
Hello. When I add prebuilt tests for ``mulhu``, for example, it causes this:
```
/home/yauheni/programs/mips/mipt-mips/simulator/risc_v/t/riscv64_test.cpp:1260:44: error: left shift count >= width of type [-Werror=shift-count-overflow]
   TEST_RR_SRC1_EQ_DEST( 8, mulhu, 143, 13<<32, 11<<32 )
```
this solves by adding ``UL`` : ``13UL<<32``.
But in case that in parent repo there is now information about it, I am not sure is it a bug or my fault?
Thank you.

pavelkryukov, 2019-04-18:
Your fix is correct. The original repo is not a C++ code, so they may have these glitches.

closed 2019-04-18 by pavelkryukov