MIPT-ILab/mipt-mips #997: Describe simulation of BP and BTB pipelines

pavelkryukov, 2019-05-02:
We have a detailed description of [BPU model](https://github.com/MIPT-ILab/mipt-mips/wiki/BPU-model), but it does not cover interaction with pipeline and process of flush.

New manual should answer these questions:
1. What are the 3 branch types?
2. What prediction should be made for the each of branch type?
3. What are the stages where prediction may be verified?
4. What is performed on branch misprediction?
5. What signals are distributed between stages to maintain the behavior?

pavelkryukov, 2019-05-05:
@BEANefiT **Disclaimer: you have 10 points already in this term. Any additional work will not affect your your course result (however, in the contribution score table all the points will be mentioned) .**

BEANefiT, 2019-05-05:
> @BEANefiT **Disclaimer: you have 10 points already in this term. Any additional work will not affect your your course result (however, in the contribution score table all the points will be mentioned) .**

Thank you for the disclaimer. I just want to try myself in explaining some things to others, that's why I've assigned this issue.

BEANefiT, 2019-05-06:
I've attempted to explain some basics of this simulation. [Here](https://github.com/MIPT-ILab/mipt-mips/wiki/Interaction-between-BPU-and-pipeline) it is.

pavelkryukov, 2019-05-06:
Your manual requires a lot of pictures,

pavelkryukov, 2019-05-06:
https://github.com/MIPT-ILab/mipt-mips/wiki/github-markdown-bkms#how-to-upload-a-picture

pavelkryukov, 2019-05-06:
1. Have you read this: https://github.com/MIPT-ILab/mipt-mips/wiki/How-to-write-Wiki-manuals ?
2. It's very hard to read about all 3 types of branchex mixed. Can we process type by type?
  * Each section for each branch should start with the topology
3. What do these pentagons mean?
4. What should I understand from that picture? https://github.com/MIPT-ILab/mipt-mips/wiki/Interaction-between-BPU-and-pipeline-pictures/pic_1.png
5. You should not copy-paste C++ code — if we wanted read C++ code, we would go to repository. You should explain _what_ it does and, more imporantly, _why_ it does it. 
6. Introduction section does not introduce to anywhere — it just says that BPU is a BPU. There is no logical connection to the next paragraph!
7. https://github.com/MIPT-ILab/mipt-mips/wiki/Interaction-between-BPU-and-pipeline#fetch-stage — why should I know all that information? Shall it be actual if we add one more source of target?

BEANefiT, 2019-05-07:
I've begun refactoring. Will finish this evening. Could you please have a quick look at it just to notice some general mistakes?

pavelkryukov, 2019-05-07:
> I've begun refactoring. Will finish this evening. 

Please do not rush — write permission won't disappear after the deadline.

BEANefiT, 2019-05-07:
> > I've begun refactoring. Will finish this evening.
> 
> Please do not rush — write permission won't disappear after the deadline.

No, I mean I'll continue working on it this evening.

pavelkryukov, 2019-05-07:
See my comments inlined on the Wiki page:
https://github.com/MIPT-ILab/mipt-mips/wiki/Branch-prediction-pipelines

I suggest to stop writing the manual now. Please provide the structure of your story instead — the lack of the skeleton would make refactoring infinite.

BEANefiT, 2019-05-07:
I've attempted to shortly introduce my view of structure. Pictures and formatting are ignored for now.

pavelkryukov, 2019-05-12:
> # Pipeline modeling

That should actually be an introduction.
I advise to put a simple picture for these three stages without any branch prediction data, like it is in our slides: fetch (and multiplexer between PC + 4 and target), decode (mentioning that it is reads target from branch encoding) and execution.

Then I suggest to write about branches performance problem — that you have to multiplex on Fetch, but control signal is ready only on late stages.

Then branch prediction is naturally introduced, we explain here why we need it. Why we need a direction prediction? Why we need a target prediction?

> I've chosen the idea of deductive method of explanation, i.e., I'll describe pipeline from tail to head. In my opinion, this will be clearer for readers.

Please no. You need to have Tarantino's talent to shoot Pulp Fiction.

I advise to start with indirect jumps. They are the simplest — predict target, calculate target, check target.

Then should go direct jumps — predict target, decode target, check target.

Then should go branches — predict direction, predict target (if branch is predicted taken), check target (if branch is predicted taken), calculate direction, check direction.

BEANefiT, 2019-05-15:
Hello. I didn't forget about this task. I just want to take a several days break, because I have some stuff to complete till the end of this term.

pavelkryukov, 2019-05-16:
Sure.

Just a reminder — you wanted to clean up that code:

https://github.com/MIPT-ILab/mipt-mips/blob/a1762774254bd4484415053c685e1b3faf17611a/simulator/modules/decode/decode.cpp#L121-L126

pavelkryukov, 2019-08-25:
@BEANefiT The manual is still actual for us. Could you please spend some time to complete it?

BEANefiT, 2019-10-14:
> @BEANefiT The manual is still actual for us. Could you please spend some time to complete it?

Hello, just saw this reply and it reminded me of this manual. Sorry for inconveniences, will refresh the work soon

pavelkryukov, 2019-12-02:
Should we expect this or it's better to put it back to the pool of tasks?

BEANefiT, 2019-12-05:
> Should we expect this or it's better to put it back to the pool of tasks?

If there is someone who wants to take this task, then it's ok to put it back to the pool. Nevertheless, I will have much more time in the end of January. That was awkward, sorry :-)