MIPT-ILab/mipt-mips #1140: Add RISC-V exception handler for MARS Kernel

pavelkryukov, 2019-11-15:
Currently our MARS kernel mode handles expections by loading MIPS32 code into memory.
https://github.com/MIPT-ILab/mipt-mips/blob/49eb1422ae74f86311473800c46ab226efc9d684/simulator/kernel/mars/mars_kernel.cpp#L234

The MIPS32 code is generated from MARS Assembler:
https://github.com/MIPT-ILab/mipt-mips/blob/master/kernels/exceptions.s

Your goal is to re-write the exception handling code for RISC-V and load it depending on modeled ISA.

pavelkryukov, 2020-02-03:
@exucutional Eric, could you please share your plan regarding this assignment?

exucutional, 2020-02-04:
Firstly I'll write riscv-driver to handle traps in the same manner as mips-driver
```c++
class DriverRISCV32 : public Driver
{
public:
    explicit DriverRISCV32( Simulator* sim) : cpu( sim) { }
    Trap handle_trap( const Operation& instr) const final 
    {
        auto trap = instr.trap_type();
        if ( trap == Trap::NO_TRAP || trap == Trap::HALT)
            return trap;

        auto tvec = cpu->read_csr_register( "mtvec");
        cpu->write_csr_register( "mcause", ...);
        cpu->write_csr_register( "mepc", ...);
        cpu->set_pc( ...);
        return Trap( Trap::NO_TRAP);
    }
    std::unique_ptr<Driver> clone() const final { return std::make_unique<DriverRISCV32>( cpu); }
private:
    Simulator* const cpu;
};
```
Also I'll implement ```get_isa()``` method for https://github.com/MIPT-ILab/mipt-mips/blob/1ccc8f8e09774f3dc5d2dad86a4566de14462b10/simulator/simulator.h#L29 to let MARS kernel knows connected simulator's isa.

What should the handler do? Without #920 it cannot even print information about exception. It can extract exception code, make some checks, clear CSRs and return. 
After implementing RISC-V syscalls, the functionality of handler may be expanded.

pavelkryukov, 2020-02-04:
Looks good.

>  Without #920 it cannot even print information about exception.

The idea is that we use MARS kernel and not RISC-V PK.
I/O will be handled by MARS kernel similarly to MIPS.

pavelkryukov, 2020-02-04:
> Firstly I'll write riscv-driver to handle traps in the same manner as mips-driver

I suggest to start with unit tests.
For MIPS, I had test binaries which I used to verify correctness — here they are likely to appear after the task completion.

pavelkryukov, 2020-02-28:
Do I understand correctly the task is close to completion?

If yes, please provide an example of interactive RISC-V program (welcome message, input two numbers, print the sum – would be enough) and describe interfaces here: https://github.com/MIPT-ILab/mipt-mips/wiki/MARS-syscalls

exucutional, 2020-03-01:
> If yes, please provide an example of interactive RISC-V program

Where should I put the code? 
Also I am trying to figure out where is the bug in performance simulator. Program works only with `-f` option.
```Performance simulator and functional simulator executed different instructions
Usually it indicates a bug in performance simulator
The different instructions are:	
Checker output: 0x1008c: {6}	addi $t1, $sp, 0	 [ $t1 = 0x1 ]
PerfSim output: 0x1008c: {6}	addi $t1, $sp, 0	 [ $t1 = 0x5 ]
```

pavelkryukov, 2020-03-01:
>  where is the bug in performance simulato

Let's focus on functional simulation. I'll check performance simulation later.

pavelkryukov, 2020-03-01:
> Where should I put the code?

I think we may have a dedicated repo: https://github.com/MIPT-ILab/riscv-mars-examples

closed 2020-03-05 by pavelkryukov

pavelkryukov, 2020-03-05:
@exucutional I appreciate the way you completed this task, with good balance of independency, involvement, and quality. Please keep that level and do not forget to move forward!

commit d4d63dbf (2020-03-05): Add RISC-V exception handler for MARS kernel (#1247)

Fixes #1140
commit 5398b077 (2020-03-05): Add RISC-V exception handler for MARS kernel (#1247)

Fixes #1140