MIPT-ILab/mipt-mips #1257: System calls should flush pipeline

pavelkryukov, 2020-03-06:
@exucutional reported in: https://github.com/MIPT-ILab/mipt-mips/issues/1140#issuecomment-593133742

> Also I am trying to figure out where is the bug in performance simulator. Program works only with -f option.

```c++
Usually it indicates a bug in performance simulator
The different instructions are:	
Checker output: 0x1008c: {6}	addi $t1, $sp, 0	 [ $t1 = 0x1 ]
PerfSim output: 0x1008c: {6}	addi $t1, $sp, 0	 [ $t1 = 0x5 ]
```

The problem existed even before RISC-V implementation. The issue is that pipeline is not flushed after system-call zero time execution. WIP to model it correctly.

exucutional, 2020-03-07:
This method is private. Should I write a test?

pavelkryukov, 2020-03-07:
You should, but you cannot write a test for private method.

First of all, you should write a test that fails because of the wrong behavior. ("red stage")
Then, you fix the code so test passes and other tests are not affected. ("green stage").
Finally, you refactor and optimize code ("refactor stage").

For more details on TDD, please follow our Wiki page: https://github.com/MIPT-ILab/mipt-mips/wiki/Introduction-to-Test-Driven-Development. It's not the best manual, but you may look for lots of resources over the Internet.

For testing, there can be two approaches:
 * Difficult one. You write unit tests for Writeback module from scratch: writing some instructions to input ports, reading the expected output from the output ports. It goes far beyond that issue, we even need a dedicated tracker.
 * Simple one (I suggest going here). We test PerfSim as a whole system. The idea is to provide a binary input and a keyboard input which cause a error. I assume the binary input has no problem for you as it is used already (see code below), keyboard input is not so hard as well, since you may use `std::instringstream` as a parameter to MARSKernel class.

https://github.com/MIPT-ILab/mipt-mips/blob/39ba2ab39d06a15191023b59ba5430db489e652b/simulator/modules/core/t/unit_test.cpp#L138-L144

closed 2020-03-17 by pavelkryukov

pavelkryukov, 2020-03-17:
Now we can study microarchitecture in a much interactive manner! Congratulations!

PR #1258 (2020-03-17): Make syscalls flush pipeline
Fixes #1257