MIPT-ILab/mipt-mips #1410: Implement ALU as a template class

graudtV, 2020-11-17:
All ALU methods are parameterized with typename I. It may be better to extract it to ALU definition, so it will look like this:
``` C++
template <typename I>
struct ALU
{
    using Predicate = bool (*)( const I*);
    // ...

    static bool eq( const I* instr)  { return instr->v_src[0] == instr->v_src[1]; }
    // ...
};
```
instead of current implementation:
``` C++
struct ALU
{
    template<typename I> using Predicate = bool (*)( const I*);
    // ...

    template<typename I> static bool eq( const I* instr)  { return instr->v_src[0] == instr->v_src[1]; }
    template<typename I> /* ... */
    // ...
};
```

pavelkryukov, 2020-11-17:
ALU has to access `I` internals, so we define it as a friend class: https://github.com/MIPT-ILab/mipt-mips/blob/a4f777b84dbb6ef1f5240246ef3a27570e0dd9f4/simulator/func_sim/operation.h#L159-L162

If I recall correctly, C++ cannot befriend with templated class, and it prevents doing that.
But I may be wrong.

pavelkryukov, 2020-11-17:
Your suggestion: https://godbolt.org/z/7h9d1a (does not work)
Current code: https://godbolt.org/z/GcP6bc (works)

pavelkryukov, 2020-11-17:
It may work if investigated more accurately. I don't find it is to be an important change actually.

pavelkryukov, 2020-11-17:
Like that: https://godbolt.org/z/oevPso

graudtV, 2020-11-17:
I suppose there are simpler ways. In this case using```execute_foo<Datapath<int>>(&instr)``` instead of ```execute_foo<Instruction<int>>(&instr)``` solves the problem.
I'll think about it more precisely although

pavelkryukov, 2020-11-17:
Your return of investment would be low, a lot of mechanical work without much value.
Consider investing your time in https://github.com/MIPT-ILab/mipt-mips/issues/304, for example

closed 2020-11-19 by pavelkryukov