MIPT-ILab/mipt-mips #1442: Problem with runing simulator with riscv-tests

artykisa, 2020-11-26:
Hello, I'm using riscv simualtor from mipt-mips (github.com/MIPT-ILab/) to run tests from riscv-tests(github.com/MIPT-ILab/riscv-tests ) also with the help of riscv-gnu-toolchain (github.com/riscv/riscv-gnu-toolchain). I'm doing this in --bp-mode
So my simulator works fine with p-format tests, but there are not too much instructions(like ~880). Also I've tried to run v-format tests, but they give me the same results in percents of mispredict despite various --bp-mode and tests, I dont think that it's good.
Then I've tried tests from benchmarks like qsort, so i've run them with this command from my build directory
**./mipt-mips --isa riscv64 -b ~/riscv-gnu-toolchain/riscv-tests/bencmarks/qsort.riscv** (path to test) **--bp-mode always_taken --mars**
After that i have this output

[image]

pavelkryukov, 2020-11-26:
> After that i have this output

This happens because we have not fixed #1076 yet, assuming next instruction is always PC + 4, while for compressed instructions it should be PC + 2.
You may try to disable compressed instructions in GCC/Assembler: look for `-march=rv64abcdefgh` string in Makefile and remove `c` letter.

> give me the same results in percents of mispredict despite various --bp-mode and tests,

Current assignment for MIPT students is to implement benchmarks sensitive to different branch prediction modes,
I can drop something I have in a private repository, so it won't be exposed before our deadlines :-)

nougat11, 2020-11-28:
> You may try to disable compressed instructions in GCC/Assembler: look for -march=rv64abcdefgh string in Makefile and remove c letter.

I didn't find this string in Makefile, also tried to write this as a flag, but it says that there is an error.
So maybe you can tell me in which makefile i should find that line or how to run this command with flag?
[image]

pavelkryukov, 2020-11-28:
Sorry, I was not precise, this string was a bad example.
In shot, RISC-V ISA has few sub-sets, and each subset is encoded with a single letter: A for atomic, B for bitmanip, C for compressed, etc.

To disable compressed instructions, add to this line:
https://github.com/riscv/riscv-tests/blob/58eb560a84ea2d6a392f7e75bba649a0c5946617/benchmarks/Makefile#L40

following options:
`-march=rv64im -mabi=lp64`

closed 2020-12-01 by pavelkryukov

avl-bsuir, 2020-12-02:
I will add few points that may clarify the current state. I succeeded with running those tests. Unfortunately, not all benchmarks could be compiled with proposed options. Those are benchamrks that use floating point instructions or atomic instructions. I could not find any option to exclude them from build, so I fixed Makefile a bit. You may find patch attached. 
[patch.txt](https://github.com/MIPT-ILab/mipt-mips/files/5631832/patch.txt)

Unfortunately, single test fails.
[image]