MIPT-ILab/mipt-mips #1444: Treating empty cycles in performance simulator

graudtV, 2020-12-01:
I was running some programs on the simulator and found a bit strange behaviour with empty loops.
The simulator executes only a fixed number of instructions, if there is an empty loop, and then stops without any messages. '-n' parameter in the command line call is ignored if it is greater than this fixed number.

Assembler source file (empty_cycle.s):
``` asm
.globl _start
.section .text

_start:
    j _start
```
Building and running options:
```
riscv64-unknown-elf-as -march=rv32g empty_cycle.s -o empty_cycle.o
riscv64-unknown-elf-ld -m elf32lriscv empty_cycle.o -o empty_cycle
../mipt-v -n 1000000 -I riscv32 -b empty_cycle --bp-mode always_not_taken 
```

Output:
```
instrs:     1
cycles:     38
```

Expected results (such results are obtained, if at least one instruction is added into the cycle, for example nop)
```
instrs:     1000000
cycles:     2500032
```

pavelkryukov, 2020-12-01:
Empty loops are often used as a halt indication. Here is where it is handled:

https://github.com/MIPT-ILab/mipt-mips/blob/ba8722436df93d689549b92bf0412c180dac61d4/simulator/func_sim/alu.h#L49-L53

graudtV, 2020-12-01:
Does it have any practical usage?

pavelkryukov, 2020-12-01:
No, that's why they are used to indicate simulation halt
For instance, RISC-V community tests use that convention.

graudtV, 2020-12-01:
Ok, thanks for answer

closed 2020-12-01 by graudtV

pavelkryukov, 2020-12-01:
You are welcome!