MIPT-ILab/mipt-mips #1464: Wrong  "bfp" and "pack" instruction encoding on branch "ooo_window".

ZVasilii, 2021-04-24:
RISK-V opcodes for instructions "bfp" and "pack" has changed in 0.92 bitmanip encoding and in  MIPT-ILab/riscv-opcodes/opcodes-rv32b file. The dissassembly unit tests has changed in "master" branch but not in the "ooo_window" branch. 
"ooo_window"
https://github.com/MIPT-ILab/mipt-mips/blob/85e7c471d433758c5ad7c1524d53517da1067f7a/simulator/risc_v/t/unit_test.cpp#L324-L328
"master"
https://github.com/MIPT-ILab/mipt-mips/blob/9486402aaa509bcb21d95b114018a48ddebfa43a/simulator/risc_v/t/unit_test.cpp#L98

"ooo_window"
https://github.com/MIPT-ILab/mipt-mips/blob/85e7c471d433758c5ad7c1524d53517da1067f7a/simulator/risc_v/t/unit_test.cpp#L249-L251
"master"
https://github.com/MIPT-ILab/mipt-mips/blob/9486402aaa509bcb21d95b114018a48ddebfa43a/simulator/risc_v/t/unit_test.cpp#L92

2 unit-tests are constantly failing.

pavelkryukov, 2021-04-24:
> The dissassembly unit tests have changed in "master" branch but not in the "ooo_window" branch.
"ooo_window"

Yes, and it is intended, because `ooo_window` branch uses 0.91 version of submodule.
Therefore, unit tests must be in sync in this version.

pavelkryukov, 2021-04-24:
Could you please clarify what is the issue you face?

closed 2021-04-24 by pavelkryukov